RPi-Distro/pi-gen #800: How to version control pi-gen customizations?

cmorganBE, 2024-10-18:
It looks like build.sh has to run from the pi-gen directory. How does one version control their changes to pi-gen, the config file etc?

Is the approach to add pi-gen to a git repo as a submodule, then run a script to copy files from the git repo into the pi-gen subdirectory, then cd pi-gen and ./build-docker.sh?

An alternative idea would be to fork pi-gen and commit changes there but that seemed slightly less clean than the copying approach.

Is there a third better option?

jayofelony, 2024-11-25:
I added both 32/64 bit pi-gen as a submodule to one of my repo's. Works great.

SvenFinn, 2025-03-18:
I have pi-gen as a submodule, with a custom script. This script combines both mine and the pi-gen repository, such that my files work as "patches", replacing the respective files and folders of pi-gen.