RPi-Distro/pi-gen #804: arm64/aarch64: export-image: 02-set-sources: QEMU OOM with message "getline (12: Cannot allocate memory)"

FacuM, 2024-11-03:
The emulator doesn't seem to be configured properly and runs out of memory when building the latest version of the `arm64` branch throwing an OOM error with the following message:

```
[19:30:08] Begin /pi-gen/export-image/02-set-sources
[19:30:08] Begin /pi-gen/export-image/02-set-sources/01-run.sh
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 https://download.docker.com/linux/debian bookworm InRelease [43.3 kB]
Get:3 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]
Get:5 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:6 http://deb.debian.org/debian bookworm/main arm64 Packages [8689 kB]
Get:7 https://download.docker.com/linux/debian bookworm/stable arm64 Packages [31.1 kB]
Get:8 http://deb.debian.org/debian bookworm/main armhf Packages [8499 kB]
Get:9 http://deb.debian.org/debian bookworm/main Translation-en [6109 kB]
Get:10 http://deb.debian.org/debian bookworm/contrib arm64 Packages [45.7 kB]
Get:11 http://deb.debian.org/debian bookworm/contrib armhf Packages [42.9 kB]
Get:12 http://deb.debian.org/debian bookworm/contrib Translation-en [48.8 kB]
Get:13 http://deb.debian.org/debian bookworm/non-free arm64 Packages [75.8 kB]
Get:14 http://deb.debian.org/debian bookworm/non-free armhf Packages [55.9 kB]
Get:15 http://deb.debian.org/debian bookworm/non-free Translation-en [67.0 kB]
Get:16 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5832 B]
Get:17 http://deb.debian.org/debian bookworm/non-free-firmware armhf Packages [5480 B]
Get:18 http://deb.debian.org/debian bookworm/non-free-firmware Translation-en [20.9 kB]
Get:19 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [187 kB]
Get:20 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [187 kB]
Get:21 http://deb.debian.org/debian-security bookworm-security/main Translation-en [116 kB]
Get:22 http://deb.debian.org/debian-security bookworm-security/contrib Translation-en [372 B]
Get:23 http://deb.debian.org/debian-security bookworm-security/non-free-firmware Translation-en [472 B]
Get:24 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [2468 B]
Get:25 http://deb.debian.org/debian bookworm-updates/main armhf Packages [2464 B]
Get:26 http://deb.debian.org/debian bookworm-updates/main Translation-en [2928 B]
Get:27 http://deb.debian.org/debian bookworm-updates/contrib arm64 Packages [768 B]
Get:28 http://deb.debian.org/debian bookworm-updates/contrib Translation-en [408 B]
Get:29 http://deb.debian.org/debian bookworm-updates/non-free armhf Packages [492 B]
Get:30 http://deb.debian.org/debian bookworm-updates/non-free arm64 Packages [12.0 kB]
Get:31 http://deb.debian.org/debian bookworm-updates/non-free Translation-en [7744 B]
Get:32 http://deb.debian.org/debian bookworm-updates/non-free-firmware arm64 Packages [616 B]
Get:33 http://deb.debian.org/debian bookworm-updates/non-free-firmware Translation-en [384 B]
Reading package lists...
E: Could not read from /var/lib/apt/lists/partial/archive.raspberrypi.com_debian_dists_bookworm_InRelease - getline (12: Cannot allocate memory)
E: The repository 'http://archive.raspberrypi.com/debian bookworm InRelease' provides only weak security information.
[19:33:14] Build failed

real    136m42.661s
user    0m0.359s
sys     0m0.426s
```

XECDesign, 2024-11-04:
Out of curiosity, what's the host OS you're using? I'm guessing it's a rather old release of Debian or Ubuntu?

When you ran `build.sh` you should've seen "WARNING: Only a native build environment is supported."

There have been enough problems with older versions of qemu that we can't guarantee it will work without issues.

closed 2024-11-06 by XECDesign

FacuM, 2024-11-06:
> Out of curiosity, what's the host OS you're using? I'm guessing it's a rather old release of Debian or Ubuntu?
>
> When you ran `build.sh` you should've seen "WARNING: Only a native build environment [...]

I'm sorry, I totally forgot to get back to you. I'm guessing it's just an edge-case kinda bug in QEMU. I'll try again later and reopen if needed with all the information about the host OS as requested.

Best regards.

nitrate92, 2025-02-27:
I've run into a similar issue using `build-docker.sh` under WSL (Ubuntu 24.10). Unless this is definitively a QEMU problem, I don't think we should be told to kick dirt for not building on ARM64.