RPi-Distro/pi-gen #809: Question about user password

D1987, 2024-11-20:
I build custom_stage via gitlab pipeline. In my a config file I pass user and password values via jinja2
```
FIRST_USER_NAME="{{ PI_USER }}"
FIRST_USER_PASS="{{ PI_PASSWORD }}"
```
From a cat command I see values applied. Then I write a final image via Raspberry Pi Imager and when I raspberry device boot but my  password wrong. It took my previous test password - pi. 
How to manage password?

XECDesign, 2024-11-21:
Are you sure your SD card hasn't gone into read-only mode?

Some SD cards fail in a way that makes them behave as if they're writing data, but if you read the data back (with cleared caches), then you get back old data. That makes the card appear as if everything is working fine, but changes don't survive reboot or remount.

Are you applying any customisation options using Imager?

XECDesign, 2024-11-22:
> I mean the settings from pi-gen build not Raspberry Imager.

I understand that, but I mean in addition to the pi-gen customisations, what are you doing with rpi-imager? It asks you whether you'd like to apply customisations and you need to answer yes or no, there's no default. If you accidentally clicked yes, it may be writing setting you had saved earlier. So I just wanted to be sure that that didn't happen.

I know it's unlikely, but  the behaviour you're seeing is a bit strange - there is no where else for it to be getting the password 'raspberry' from.