RPi-Distro/pi-gen #814: Set fixed MAC for USB gadget mode at first boot

jayofelony, 2024-12-02:
When a RPi zero (2) is being connected to a pc it will get a new random mac address each time.
When ifupdown was removed it got trivial to get g_ether working again, so I installed it again which made everything work again. 
https://github.com/RPi-Distro/pi-gen/issues/810

But it would be preferable if g_ether got a fixed MAC address at first boot. I was trying to add the following to `/usr/lib/raspberrypi-sys-mods/firstboot`:

`echo 'options g_ether host_addr='$(dmesg | awk '/: HOST MAC/{print $NF}')' dev_addr='$(dmesg | awk '/: MAC/{print $NF}') > etc/modprobe.d/g_ether.conf`

Without any luck, it didn't work. As I am still new at bash, so I am probably doing something wrong. Some guidance would be nice.

XECDesign, 2024-12-02:
You may have better luck on the [forum](https://forums.raspberrypi.com/), since this isn't really a pi-gen issue.

In `firstboot`, `/` is mounted as read-only for most of the time, so where you're adding that line may make a difference. Also it may be running before the `g_ether` module is loaded, so the lines you're expecting may not have appeared in dmesg yet. Instead of trying to get the addresses from dmesg, I'd derive them form the device serial number in some way.

At some point `firstboot` should disappear altogether in favour of initramfs hooks [like this one](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/initramfs-tools/scripts/local-bottom/firstboot_fstrim). So that may be a better approach here as well.

jayofelony, 2024-12-02:
Ok, thank you. Appreciate you taken the time to reply. I know it isn't a pi-gen issue, I've asked also on rpi stackexchange but the people there are usually kind of jerks. I'll close it.

closed 2024-12-02 by jayofelony