RPi-Distro/pi-gen #815: Build via Docker in GitHub actions.

pompushko, 2024-12-08:
Hello

Does someone try to do that? Maybe someone can share yaml?  I see that it possible via GitLab...

Thank you

pompushko, 2024-12-09:
Well
Here is my realization or [here](https://github.com/RPi-Distro/pi-gen/issues/785)

```
name: Build Docker Image

on:
  workflow_dispatch:
  push:
    branches:
      - main
    paths-ignore:
      - '.github/**'
  pull_request:
    branches:
      - main
    paths-ignore:
      - '.github/**'

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
      - name: Checkout repository
        uses: actions/checkout@v4

      - name: Install dependencies
        run: |
          sudo apt-get update
          sudo apt-get install -y coreutils quilt parted qemu-user-static debootstrap zerofree zip dosfstools libarchive-tools libcap2-bin grep rsync xz-utils file git curl bc gpg pigz xxd arch-test

      - name: Build Docker image
        run: |
          chmod +x ./build-docker.sh
          ./build-docker.sh

      - name: Upload logs
        if: success()
        uses: actions/upload-artifact@v4
        with:
          name: build-logs
          path: deploy/build-docker.log
          retention-days: 7
          compression-level: 9

      - name: Authenticate to Google Cloud
        id: auth
        uses: google-github-actions/auth@v2
        with:
          credentials_json: ${{ secrets.GCR_SA_KEY }}

      - name: 'Set up Cloud SDK'
        uses: 'google-github-actions/setup-gcloud@v2'
        
      - name: 'Use gcloud CLI'
        run: 'gcloud info'

      - name: 'Upload image to GCP bucket'
        uses: 'google-github-actions/upload-cloud-storage@v2'
        with:
          path: 'deploy'
          destination: ${{ vars.GS_BUCKET }}
          glob: '*.zip'
          process_gcloudignore: false
          parent: false
```

closed 2024-12-09 by pompushko

usimd, 2025-01-12:
Fwiw, [usimd/pi-gen-action](https://github.com/usimd/pi-gen-action) accepts forks from the original pi-gen repo for that very use case:

https://github.com/usimd/pi-gen-action/blob/7c87fdbda7981eb48776d52230f25f2d850ec82e/action.yml#L111-L114