RPi-Distro/pi-gen #818: Patches don't seem to be applied correctly

montmejat, 2024-12-14:
I have a custom stage with a patch:

```
mycustomstage/
├── 00-substage1
│   └── 00-run-chroot.sh
├── 01-substage2
│   ├── 00-packages
│   ├── 01-patches
│   │   ├── EDIT
│   │   ├── Static-ethernet.nmconnection.patch
│   │   └── series
│   ├── 02-run.sh
│   ├── 03-run-chroot.sh
│   ├── 04-run-chroot.sh
│   └── 05-run-chroot.sh
├── EXPORT_IMAGE
└── prerun.sh
```

with:

```
--- /dev/null
+++ /etc/NetworkManager/system-connections/Static-ethernet.nmconnection
@@ -0,0 +1,19 @@
+[connection]
+id=Static-ethernet
+uuid=77201d0b-7cb9-4fc4-8d40-2b2b794f69ac
+type=ethernet
+interface-name=eth0
+timestamp=1733930583
+
+[ethernet]
+
+[ipv4]
+address1=192.168.0.101/24
+method=manual
+
+[ipv6]
+addr-gen-mode=default
+method=disabled
+
+[proxy]
+
```

However, it keeps on getting patched in `pi-gen/work/mycustomstage/etc/...` instead of `pi-gen/work/mycustomstage/rootfs/etc/...`! What did I miss here? :)

I've also tried:

```
+++ /etc/NetworkManager/system-connections/Static-ethernet.nmconnection
```

```
+++ etc/NetworkManager/system-connections/Static-ethernet.nmconnection
```

```
+++ rootfs/etc/NetworkManager/system-connections/Static-ethernet.nmconnection
```

Thanks!

orontee, 2024-12-15:
According to https://github.com/RPi-Distro/pi-gen/blob/arm64/build.sh#L41, `quilt` applies patches while current directory is `STAGE_WORK_DIR`, so you should have a `/rootfs` prefix in your patches if they're expected to modify the stage root filesystem.

montmejat, 2024-12-15:
Ooh okay, I would have expected `+++ rootfs/etc/...` to work since `build.sh` changes the current directory to `pi-gen/work/mycustomstage`.  I noticed that quilt just ignores everything before the first `/` (`+++ whatever/etc/...` has the same effect), I'm not sure why... :thinking: Anyways, thanks!

closed 2024-12-15 by montmejat