RPi-Distro/pi-gen #824: Question about overlaying certain stage

BetelGeuseee, 2025-01-24:
**What I am doing:**

I created a custom stage3 directory where i included a binary executable that should be copied somewhere in the rootfs. I modified the 01-run.sh script. Also i need some directories to be created while building stage3, but in the final image these changes are not reflected.

Is there any standard way that i should know or follow for overlaying my custom stage?

XECDesign, 2025-01-24:
Do you have an example of what you've done and what's not working?

hugoj-kheops, 2025-05-21:
I think I experience the same issue.
I use the `build-docker.sh` script on the bullseye branch.

I want to create two images based on a different stage list manipulated on CI:
- one for production
- one to compile my production Python Pyinstaller program, to be deployed on the first image (to have the appropriate OS dependencies required for the python dependencies)

Two issues I experienced:

1. While trying to install PDM using `on_chroot` script, it is logged properly but on the final image, no presence of PDM at all in the filesystem, even installed as root or as the "pi" user.
2. I also tried to copy a customization file to the `${ROOTFS}/etc/bluetooth/` folder (maybe not the right place but anyway) using the files/ folder and the install command, it didn't perform it in the end (I saw the stage script executed in the logs but no file copied).