RPi-Distro/pi-gen #827: Boot pi-gen generated image with QEMU aarch64

Dlozitskiy, 2025-02-04:
Can someone provide tips on how to boot up a pi-gen generated image with QEMU?

I'm building the image for raspi4b, tried with and without `USE_QEMU`, however boot with qemu-system-aarch64 fails on mounting the root fs. The image works well with the device though, so something is missing in the way I run qemu-system-aarch64.

QEMU emulator version is 9.2.0

Here is my run command:

```
/usr/local/bin/qemu-system-aarch64 \
    -machine raspi4b \
    -cpu cortex-a72 \
    -smp 4 \
    -m 2G \
    -kernel kernel8.img \
    -append "root=/dev/mmcblk0p2 rootfstype=ext4 rw console=ttyAMA0,115200 loglevel=8 earlycon=pl011,0xfe201000 earlyprintk panic=1" \
    -drive format=raw,file=2025-02-04-raspios-bookworm-armhf-lite-qemu.img,if=sd \
    -nographic \
    -device usb-net,netdev=net0 \
    -netdev user,id=net0,hostfwd=tcp::5022-:22
```

Tried both on Ubuntu and MacOS with the same result.

```
[    2.475523] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    2.476969] CPU: 3 PID: 1 Comm: swapper/0 Not tainted 6.6.74+rpt-rpi-v8 #1  Debian 1:6.6.74-1+rpt1
[    2.479543] Hardware name: Raspberry Pi 4 Model B (DT)
[    2.481338] Call trace:
[    2.482583]  dump_backtrace+0xa0/0x100
[    2.483679]  show_stack+0x20/0x38
[    2.485203]  dump_stack_lvl+0x48/0x60
[    2.486026]  dump_stack+0x18/0x28
[    2.486883]  panic+0x330/0x398
[    2.487284]  mount_root_generic+0x260/0x348
[    2.488042]  mount_root+0x16c/0x2e8
[    2.488452]  prepare_namespace+0x74/0x2b8
[    2.490061]  kernel_init_freeable+0x374/0x3d8
[    2.491073]  kernel_init+0x2c/0x1f8
[    2.491694]  ret_from_fork+0x10/0x20
[    2.492673] SMP: stopping secondary CPUs
[    2.494232] Kernel Offset: disabled
[    2.494633] CPU features: 0x0,80000200,3c020000,0000421b
```

Tried different partitions /dev/mmcblk0p0-4 but that doesn't help.

bjbcan, 2025-03-05:
I build the image with QEMU and am using the same version as you on an M1 Mac.
```
qemu-system-aarch64 \
    -M raspi3b \
    -cpu cortex-a72 \
    -append "rw earlyprintk loglevel=8 console=ttyAMA0,115200 dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootdelay=1" \
    -dtb bcm2710-rpi-3-b-plus.dtb \
    -drive "if=sd,format=raw,file=$IMAGE_FILE"  \
    -kernel kernel8.img \
    -m 1G -smp 4 -nographic 
```

A less prescriptive version also works for me:
```
qemu-system-aarch64 \
	-m 1024 -M raspi3b -kernel kernel8.img \
	-dtb bcm2710-rpi-3-b-plus.dtb -sd ${IMAGE_FILE} \
	-append "console=ttyAMA0 root=/dev/mmcblk0p2 rw rootwait rootfstype=ext4" \
	-nographic 
```

You can get the device tree blob file from any RPi image at `/boot/firmware` just like you got the kernel image. Note that I'm not able to get any network device configured as the QEMU installation I have doesn't have any nic models built in for RPis.  
```
qemu-system-aarch64 -M raspi3b -nic model=help
Available NIC models for this configuration:
<blank>
```

TimJDFletcher, 2025-05-14:
I've been looking at this recently the problem is that for machine type raspi4b the SD card is detected as mmcblk1 not 0, the fix for you is to change the append line to: `rootfs=/dev/mmcblk1p2`