RPi-Distro/pi-gen #829: Docker build on x86-64 want 00-run.sh scripts to be able to build and install arm64 packages

chmorgan, 2025-02-10:
arm64 branch on x86-64

For a custom run script we'd like to avoid impacting the final image with build-only packages:

01-run.sh

```bash
apt install -y a-dev b-dev c-dev d-dev e-dev f-dev g-dev
wget package
build package
install to ${ROOTFS_DIR}
```

The problem is that this is that for x86-64 hosts the 01-run.sh runs inside of a x86-64 docker container, and all of the things built there are x86-64 libraries and binaries.

We *could* perform these steps inside of the chroot, and then uninstall the various *-dev packages, although this seemed more error prone.

Is there some approach others have used that we are missing here?

The scripts today work well on aarch64 (apple silicon) or GitHub ubuntu-24.04-arm (aarch64) runners of course, but even still those are a minority. Would like this to work across aarch64 and more common x86-64 systems.

XECDesign, 2025-02-10:
This is all a bit outside the scope of what this issue tracker is for, but if you really wanted to do it this way, you could cross-compile the application. But if it's something that requires a lot of `-dev` packages, it's probably non-trivial.

If I was doing this, I wouldn't be building the software as a part of the image build. I'd pre-build it as a debian package as a part of a separate process and install it as a normal apt package.

closed 2025-02-10 by XECDesign

chmorgan, 2025-02-10:
I appreciate the response. If the answer is “the whole idea is wrong, do
this instead” then at least there is a direction to head based on someone
else with more experience here.

Are there any examples of this that you know of that I could reference?

XECDesign, 2025-02-10:
I wouldn't go as far as to say the whole idea is wrong. If it works for your use case, then there's nothing wrong with it.

I think the 'official' response would be, you need to run it natively rather than on amd64. Then the problem goes away anyway. But I understand that's not always ideal.

You can do what you're doing already, but build inside the chroot. That's the simplest option.

Or, you can pre-build and host the deb elsewhere, then include the repo in pi-gen and install it through a `packages` file. This is probably the most 'correct' approach, but would be the trickiest if you're not already familiar with Debian package and repo management.

What are you leaning towards and how comfortable are you working with Debian packages? Do you have a link to the software you're building or is it something private?

chmorgan, 2025-02-10:
Works vs good idea can have a wide gap :-)

This is more than enough help, we already build Debian packages for the
main application. It’s two additional packages and we can use salsa for
bluez, the other one we’ll have to hand craft maybe.

Thank you again for the pointer in the right direction.

XECDesign, 2025-02-10:
> Works vs good idea can have a wide gap :-)

Ha yeah, I'm just used to people being reluctant to cross that gap.

> Thank you again for the pointer in the right direction.

No worries and good luck with it.

Depending on what you're doing, it may also we worth taking a look at https://github.com/raspberrypi/rpi-image-gen