RPi-Distro/pi-gen #832: bullseye build error with qemu: uncaught target signal 11 (Segmentation fault) - core dumped

hcwinsemius, 2025-03-12:
For reasons of backwards compatibility, I need to build a bespoke bullseye-based image. The build process crashes within `stage0/rootfs` as follows (I replaced folder names for `$PIGEN`):

```
I: Extracting zlib1g...
W: Failure trying to run: chroot "$PIGEN/pi-gen/work/raspios-bullseye-arm64/stage0/rootfs" /sbin/ldconfig
W: See $PIGEN/pi-gen/work/raspios-bullseye-arm64/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '$PIGEN/pi-gen/work/raspios-bullseye-arm64/stage0/rootfs/debootstrap': Directory not empty
[13:34:55] bootstrap failed: please check $PIGEN/pi-gen/work/raspios-bullseye-arm64/stage0/debootstrap.log
```

`debootstrap.log` contains:

```
2025-03-12 13:34:51 URL:http://deb.debian.org/debian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2+deb11u2_arm64.deb [88168/88168] -> "$PIGEN/pi-gen/work/raspios-bullseye-arm64/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2+deb11u2_arm64.deb" [1]
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
```
Further info:
* I have qemu 8.2.2 (default for current LTS of Ubuntu). 
* Setting `USE_QEMU=0` does not help.
* Trying to build with docker also does not help. Exactly the same error.

I saw earlier posts similar to this, but quite old and likely much outdated. Any chance you can help out?

Thanks!

XECDesign, 2025-03-13:
I think you'd have to build natively on an arm64 machine. I can confirm I'm not having any luck on Ubuntu 24.04 either and I'm 99% certain the issue is the version of qemu.

XECDesign, 2025-03-13:
Also, heads up we'll be dropping support for Bullseye whenever we have Trixie ready which may be as early as a few months from now.

gxanshu, 2025-03-25:
is there any way to build raspberry pi image on x64 system for arm64 ? 

i'm getting this error

```
arm64: ok
[15:01:09] Begin /pi-gen
[15:01:09] Begin /pi-gen/stage0
[15:01:09] Begin /pi-gen/stage0/prerun.sh
[15:01:09] End /pi-gen/stage0/prerun.sh
[15:01:09] Begin /pi-gen/stage0/00-configure-apt
[15:01:09] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists.../bin/bash: line 7:   127 Segmentation fault      apt-get update
[15:01:14] Build failed
```

XECDesign, 2025-03-25:
You'd need to experiment with building different versions of qemu-user and find one that works.

closed 2025-03-25 by XECDesign