RPi-Distro/pi-gen #836: Reboot probability stuck

Shanks0815, 2025-04-02:
We found that repeated reboots would occasionally hang at the shutdown logo stage, taking approximately eight minutes to complete the reboot.

1. Tested on pi4 with 2024-11-19-raspios-bookworm-arm64.img.
2. Used a script to do reboot and log the time，no more actions.

ghollingworth, 2025-04-02:
Can you confirm your test conditions?

How did you install the image?
Did you pre-set configuration from Raspberry Pi Imager or did you follow through the Wizard?
What settings did you make (WiFi configuration, ssh etc.) it's possible the setting up (or not) of the configuration will be important.

XECDesign, 2025-04-02:
When it's stuck like that, you may also be able to press ESC to hide the splash screen and see more output. You should be able to see something like "Waiting for..." and some service. This may happen when a process doesn't terminate when asked. If you let us know what service that is, we may be able to dig in a bit more.

Shanks0815, 2025-04-03:
We downloaded [image](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2024-11-19/2024-11-19-raspios-bookworm-arm64.img.xz) from website and unzipped it.
No configuration has been preset from the Raspberry Pi imager.

> When it's stuck like that, you may also be able to press ESC to hide the splash screen and see more output. You should be able to see something like "Waiting for..." and some service. This may [...]

We'll try this way to grab more logs.

ahnniu, 2025-04-23:
@ghollingworth @XECDesign Thank you very much for your help in this issue!

We have a Raspberry Pi CM4 customer that has also reported this issue, here is the Reboot blocked log.

- [log of reboot blocked for more than 10 mins.txt](https://github.com/user-attachments/files/19866572/log.of.reboot.blocked.for.more.than.8.mins.txt)

The customer also did a loop reboot test, here is the detailed test data, there are blocked for 1 minute and blocked for 10 minutes.

| No.  | Total Time (min) | Reboot Attempts | 1-Minute Stuck Count | 10-Minute Stuck Count |
| :--: | :--------------: | :-------------: | :------------------: | :-------------------: |
|  1   |       775        |       770       |          4           |           0           |
|  2   |       792        |       758       |          3           |           3           |
|  3   |       795        |       758       |          7           |           3           |
|  4   |       775        |       762       |          1           |           1           |
|  5   |       801        |       796       |          5           |           0           |
|  6   |       774        |       769       |          5           |           0           |
|  7   |       796        |       795       |          1           |           0           |
|  8   |       787        |       763       |          4           |           2           |
|  9   |       799        |       755       |          4           |           4           |
|  10  |       779        |       766       |          3           |           1           |
|  11  |       788        |       773       |          5           |           1           |
|  12  |       788        |       638       |          2           |          13           |
|  13  |       782        |       766       |          6           |           1           |
|  14  |       784        |       758       |          1           |           1           |
|  15  |       806        |       783       |          3           |           2           |

Hopefully the logs and test data will help us do further problem localization.

Best Regards,
Paul

pelwell, 2025-04-24:
I have a Pi 4 running the current 64-bit desktop image and a script rebooting in a loop, and for the last hour it's been fine, but I do see that this is in keeping with your observed failure rates.

What would be useful is the same log output for a successful reboot, so we can see which messages have not yet appeared in the stall log above.

ghollingworth, 2025-04-24:
Also for completeness, can you give us your test script and method of reproduction, as well as the specific device that you used so we can make sure we're comparing the same things...

pelwell, 2025-04-24:
I just got a stall. It was nowhere near the kernel reboot logic - systemd had barely started shutting down.

A normal reboot looks like this:
```
raspberrypi login: [   29.332489] watchdog: watchdog0: watchdog did not stop!
[   29.574992] systemd-shutdown[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog0
[   29.586473] systemd-shutdown[1]: Watchdog running with a hardware timeout of 10min.
[   29.863866] systemd-shutdown[1]: Syncing filesystems and block devices.
[   30.120837] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   30.136036] systemd-journald[274]: Received SIGTERM from PID 1 (systemd-shutdow).
[   30.183303] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   30.196314] systemd-shutdown[1]: Process 1448 (plymouthd-fd-es) has been marked to be excluded from killing. It is running from the root file system, and thus likely to block re-mounting of the root file system to read-only. Please consider moving it into an initrd file system instead.
[   30.222752] systemd-shutdown[1]: Unmounting file systems.
[   30.229614] (sd-remount)[1450]: Remounting '/' read-only with options ''.
[   30.282022] EXT4-fs (mmcblk0p2): re-mounted a36be96c-66be-4487-a7a6-0481bca99d89 ro. Quota mode: none.
[   30.300152] systemd-shutdown[1]: All filesystems unmounted.
[   30.305854] systemd-shutdown[1]: Deactivating swaps.
[   30.311023] systemd-shutdown[1]: All swaps deactivated.
[   30.316369] systemd-shutdown[1]: Detaching loop devices.
[   30.322943] systemd-shutdown[1]: All loop devices detached.
[   30.328619] systemd-shutdown[1]: Stopping MD devices.
[   30.333913] systemd-shutdown[1]: All MD devices stopped.
[   30.339321] systemd-shutdown[1]: Detaching DM devices.
[   30.344657] systemd-shutdown[1]: All DM devices detached.
[   30.350142] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[   30.359955] watchdog: watchdog0: watchdog did not stop!
[   30.374824] systemd-shutdown[1]: Syncing filesystems and block devices.
[   30.381678] systemd-shutdown[1]: Rebooting.
[   30.385932] rpi_firmware_notify_reboot(0)
[   30.558586] kvm: exiting hardware virtualization
[   30.563299] reboot: Restarting system
```
But this one was:
```
raspberrypi login: [  119.445991] watchdog: watchdog0: watchdog did not stop!
[  119.528651] systemd-shutdown[1]: Using hardware watchdog 'Broadcom BCM2835 Watchdog timer', version 0, device /dev/watchdog0
[  119.540157] systemd-shutdown[1]: Watchdog running with a hardware timeout of 10min.
[  119.898010] systemd-shutdown[1]: Syncing filesystems and block devices.
[  120.005947] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  120.019723] systemd-journald[271]: Received SIGTERM from PID 1 (systemd-shutdow).
[  120.068560] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  120.081563] systemd-shutdown[1]: Process 1435 (plymouthd-fd-es) has been marked to be excluded from killing. It is running from the root file system, and thus likely to block re-mounting of the root file system to read-only. Please consider moving it into an initrd file system instead.
[  120.108112] systemd-shutdown[1]: Unmounting file systems.
[  120.114956] (sd-remount)[1437]: Remounting '/' read-only with options ''.
[  120.167199] EXT4-fs (mmcblk0p2): re-mounted a36be96c-66be-4487-a7a6-0481bca99d89 ro. Quota mode: none.
[  120.185255] systemd-shutdown[1]: All filesystems unmounted.
[  120.191063] systemd-shutdown[1]: Deactivating swaps.
[  120.196215] systemd-shutdown[1]: All swaps deactivated.
[  120.201526] systemd-shutdown[1]: Detaching loop devices.
[  120.208094] systemd-shutdown[1]: All loop devices detached.
[  120.213770] systemd-shutdown[1]: Stopping MD devices.
[  120.219022] systemd-shutdown[1]: All MD devices stopped.
[  120.224421] systemd-shutdown[1]: Detaching DM devices.
[  120.229754] systemd-shutdown[1]: All DM devices detached.
[  120.235240] systemd-shutdown[1]: All filesystems, swaps, loop devices, MD devices and DM devices detached.
[  120.245044] watchdog: watchdog0: watchdog did not stop!
[  120.259676] systemd-shutdown[1]: Syncing filesystems and block devices.
[  120.266539] systemd-shutdown[1]: Rebooting.
[  120.270796] rpi_firmware_notify_reboot(0)
[  120.444232] kvm: exiting hardware virtualization
[  120.448944] reboot: Restarting system
```
While it was stalled, the reboot splash screen was visible with the message "plymouth-reboot.service".

I'll leave it running overnight to gather some more data, but if all the delays are like that then I'll try to understand that watchdog error message.

ghollingworth, 2025-04-24:
It would be interesting to see what happens if you uninstall plymouth...

pelwell, 2025-04-25:
Overnight I got about 10 longer reboots, but nothing more than 2 minutes. Disabling plymouth doesn't seem to make any difference.

Tha

Restarting with more systemd logging I eventually found this smoking gun:
```
[   57.546705] systemd[1]: Stopped cups.service - CUPS Scheduler.
[  139.162591] systemd[1]: session-1.scope: Stopping timed out. Killing.
[  139.169667] systemd[1]: session-1.scope: Killing process 1184 (menu-cached) with signal SIGKILL.
[  139.179017] systemd[1]: session-1.scope: Killing process 1186 (gmain) with signal SIGKILL.
[  139.190334] systemd[1]: session-1.scope: Failed with result 'timeout'.
[  139.198027] systemd[1]: Stopped session-1.scope - Session 1 of User pi.
[  139.205829] systemd[1]: session-1.scope: Consumed 11.975s CPU time.
```
And then a similar grouping later:
```
[   58.925937] systemd[1]: Stopped cups.service - CUPS Scheduler.
[  139.899067] systemd[1]: session-1.scope: Stopping timed out. Killing.
[  139.906115] systemd[1]: session-1.scope: Killing process 1176 (menu-cached) with signal SIGKILL.
[  139.915451] systemd[1]: session-1.scope: Killing process 1179 (gmain) with signal SIGKILL.
[  139.925389] systemd[1]: session-1.scope: Killing process 1180 (n/a) with signal SIGKILL.
[  139.935440] systemd[1]: session-1.scope: Failed with result 'timeout'.
[  139.943123] systemd[1]: Stopped session-1.scope - Session 1 of User pi.
[  139.952756] systemd[1]: session-1.scope: Consumed 11.819s CPU time.
```
On two other occasions the user service just seemed to take a long time to exit:
```
[   56.857184] systemd[1]: cups.service: Deactivated successfully.
[   56.863553] systemd[1]: Stopped cups.service - CUPS Scheduler.
[  146.742678] systemd[1]: user@1000.service: Deactivated successfully.
[  146.750693] systemd[1]: Stopped user@1000.service - User Manager for UID 1000.
[  146.758716] systemd[1]: user@1000.service: Consumed 2.878s CPU time.
```
and:
```
[   59.357922] systemd[1]: Stopped systemd-logind.service - User Login Management.
[  149.421857] systemd[1]: user@1000.service: Deactivated successfully.
[  149.429570] systemd[1]: Stopped user@1000.service - User Manager for UID 1000.
[  149.437593] systemd[1]: user@1000.service: Consumed 2.879s CPU time.
[  149.552802] systemd[1]: Stopping systemd-user-sessions.service - Permit User Sessions...
```
Now switching to console (no desktop) mode to see if the stalling goes away.

pelwell, 2025-04-28:
Over the weekend, running in console mode (with `fsck.repair=no` in cmdline.txt, although it probably made no difference) there were 5 slow reboots. All of the delays manifested as 90 second timeouts shutting down `cups-browsed.service`. The test is now running with the cups-browsed service disabled and the GUI, plymouth and the fsck repair re-enabled.

pelwell, 2025-04-29:
After disabling cups-browsed, plymouth and fsck.repair (perhaps not all necessary), there were only two delays overnight - and no obvious error messages:
```
[   58.955806] systemd[1]: Unmounted run-user-1000-gvfs.mount - /run/user/1000/gvfs.
[   58.965594] systemd[1]: systemd-logind.service: Deactivated successfully.
[   58.973237] systemd[1]: Stopped systemd-logind.service - User Login Management.
[   58.984413] systemd[1]: plymouth-reboot.service - Show Plymouth Reboot Screen was skipped because of an unmet condition check (ConditionKernelCommandLine=!plymouth.enable=0).
[   59.000346] systemd[1]: plymouth-switch-root-initramfs.service - Tell Plymouth To Jump To initramfs was skipped because of an unmet condition check (ConditionPathExists=/run/initramfs/bin/sh).
[  149.093192] systemd[1]: user@1000.service: Deactivated successfully.
[  149.100925] systemd[1]: Stopped user@1000.service - User Manager for UID 1000.
[  149.108876] systemd[1]: user@1000.service: Consumed 2.864s CPU time.
----
[   53.909127] systemd[1]: Unmounted run-user-1000-gvfs.mount - /run/user/1000/gvfs.
[   54.171072] systemd[1]: systemd-random-seed.service: Deactivated successfully.
[   54.178969] systemd[1]: Stopped systemd-random-seed.service - Load/Save Random Seed.
[   54.187786] systemd[1]: cups.service: Deactivated successfully.
[   54.194199] systemd[1]: Stopped cups.service - CUPS Scheduler.
[  144.065677] systemd[1]: user@1000.service: Deactivated successfully.
[  144.073307] systemd[1]: Stopped user@1000.service - User Manager for UID 1000.
[  144.081248] systemd[1]: user@1000.service: Consumed 2.935s CPU time.
```
I would definitely recommend disabling cups-browsed.

However, I've never seen the 8+ minute delays that were reported; the only things I'm aware of that can stall for close to 10 minutes are SD/SDIO scans. If you are using a Compute Module without WiFi or a device using NVMe/USB storage and no SD card you may be hitting this problem. The latest CM5 EEPROM image will automatically disable the SDIO interface for SKUs without WiFi.

pelwell, 2025-04-30:
A (likely) final update: since also switching back to console login (i.e. no desktop), there's not been a single reboot delay in over 1200 reboots.

To summarise:
* Unless you rely on them, consider disabling cups-browsed and plymouth.
* If you are using a device with an empty but usable SD card slot where the device doesn't have a card detect signal, or if it's a Compute Module that could have onboard WiFi but doesn't, there are things you can do to disable the SD/SDIO interfaces to prevent stalls of up to 10 minutes.

pelwell, 2025-05-01:
The test has been halted after ~3000 successful reboots. In that time there was one run that didn't reach the "Restarting system" message until the 90.6s mark, the next latest being 55.6s. However, that turned out to be due to a late start, apparently with due to some sort of networking glitch.

closed 2025-05-01 by pelwell