RPi-Distro/pi-gen #853: Move resize2fs_once to native systemd unit file

martignoni, 2025-07-05:
In a trixie-updated RPiOS, getting following warnings:
````
systemd-sysv-generator[2364]: SysV service '/etc/init.d/resize2fs_once' lacks a native systemd unit file, automatically generating a unit file for compatibility.
systemd-sysv-generator[2364]: Please update package to include a native systemd unit file.
systemd-sysv-generator[2364]: ! This compatibility logic is deprecated, expect removal soon. !
```

XECDesign, 2025-07-06:
I can't think of any valid/supported configuration where you would end up with that file on a Trixie install. If you start with a bookworm image, the file deletes itself after it runs.

We haven't released any trixie images to could contain that file.

martignoni, 2025-07-06:
> We haven't released any trixie images to could contain that file.

Yes indeed. Does it imply that the trixie image would include an updated way (instead of the `resize2fs_once` init script) to resize the partition at first boot? Can you already comment about it?

martignoni, 2025-07-06:
> Let me know if that's something that would be helpful for you.

Yes, I'd appreciate it.

timsurber, 2025-07-18:
@XECDesign do you have the trixie images available already for testing?

XECDesign, 2025-07-28:
@timsurber Nothing public, no. If something becomes available, you'll be able to find it on the [Beta testing forum](https://forums.raspberrypi.com/viewforum.php?f=173).

martignoni, 2025-10-03:
> That's exactly right, yeah. It's now handled by initramfs. If you have ` resize` in your cmdline.txt, it will regenerate your Disk ID, resize the partition and the filesystem before it gets mounted. I've done it this way to eliminate the need for reboots.

Thnaks a lot for your comment @XECDesign. Now that Trixie images are released, could you point me to where to look at for understanding the new resize mecanism? I've seen this service: `rpi-resize.service`, but cannot understand how it exactly works.

I've also seen that within `raspi-config`, the old way of resizing is still there, which puzzled me. Can you comment about it?

XECDesign, 2025-10-06:
> > That's exactly right, yeah. It's now handled by initramfs. If you have ` resize` in your cmdline.txt, it will regenerate your Disk ID, resize the partition and the filesystem before it gets mounted. I've done it this way to eliminate the need for reboots.
> 
> Thanks a lot for your comments [@XECDesign](https://github.com/XECDesign). Now that Trixie images are released, could you point me to where to look at for understanding the new resize mecanism? I've seen this service: `rpi-resize.service`, but cannot understand how it exactly works.
> 
> I've also seen that within `raspi-config`, the old way of resizing is still there, which puzzled me. Can you comment about it?

I've posted a brief explanation [here](https://github.com/RPi-Distro/pi-gen/issues/861#issuecomment-3371040040).

It works by pulling in the existing `systemd-growfs-root.service` into the `Wants` field.  But the actual partition resize is done in initramfs.

There are people who act like developers make changes purely out of boredom. For that reason, I haven't touched the raspi-config option, but it doesn't get tested and I can't vouch for it working. Maybe there are still some people relying on it in some way.