RPi-Distro/pi-gen #854: failed running firstboot root partition failed

fiepfiep, 2025-07-15:
when building an image using ubuntu25.04 - images won't boot on my pi.
I get a blue screen with error message:

```
failed running firstboot root partition failed
```

XECDesign, 2025-07-15:
Any more information? Is this a stock arm64 bookworm lite image with no customisations?

Was it built on an arm64 machine or on amd64 without qemu-user?

fiepfiep, 2025-07-15:
I am building the bullseye arm image.
ubuntu is running on a x86-64 machine

XECDesign, 2025-07-15:
Alright, I'll spin up a 25.04 VM and give it a go when I get a chance.

XECDesign, 2025-07-15:
Checked the equivalent nightly image (built on debian arm64) and that boots fine. So it does seem to be a qemu or 25.04 change that's responsible. I'll take a look.

commit 14895206 (2025-07-15): Disable the orphan_file feature

orphan_file is supported by e2fsprogs 1.47.0 and later. Bullseye ships 1.46.2

Fixes #854
closed 2025-07-18 by XECDesign