RPi-Distro/pi-gen #861: resize2fs_once is gone?

CaptainMcCrank, 2025-08-19:
Any guidance on why it was removed and where it went?  I have build scripts that depend on this app.

https://raw.githubusercontent.com/RPi-Distro/pi-gen/master/stage2/01-sys-tweaks/files/resize2fs_once 404's.  was working yesterday.

Ivoz, 2025-09-25:
Not sure if it has something to do with 04bc6843a9d144e6813780afa5792d7ba6816043 adding a `resize` to cmdline?

CaptainMcCrank, 2025-09-26:
@Ivoz Nice catch- but still this is super cryptic.  I'd love clarity on a resizing strategy for fresh builds.  This change busted my build process.  What's the best way to make sure our customers can deploy firmware that expands to the max size of the customer's SD card?

XECDesign, 2025-10-06:
There's an [initramfs script to resize the partition](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/pios/trixie/initramfs-tools/scripts/local-premount/resize_early).

Then there's a systemd service ([rpi-resize.service]( https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/pios/trixie/debian/raspberrypi-sys-mods.rpi-resize.service)) that uses existing services to grow the root partition and run fstrim on first boot, then disables itself.

And [here'](https://www.freedesktop.org/software/systemd/man/latest/machine-id.html#First%20Boot%20Semantics)s how systemd determines whether you're booting for the first time.

XECDesign, 2025-10-16:
> [@XECDesign](https://github.com/XECDesign) This new rpi-resize.service does not check for "resize" in cmdline.txt and always runs.

It doesn't need to because it doesn't resize the partition.

 > In addition cloud-init "growpart" and "resizefs" also don't adhere to this check.

Yes, this is a problem cloud-init and future images will have those modules disabled.